ardlib/bosejis_xxtea #2: xxtea-iot-crypt doesnt decrypt

ThePaceMaker, 2016-11-22:
What ever you did, it doesn't conform to the xxtea standard!

If you run your example xxtea_Easy.ino, you get the following result returned...

 Password : Hello Password
 Plain Text: Hi There we can work with this
 Encrypted Data: C8F1CB5F5E7D3BC5EE2753FFC36F8B7AF96759EEF6973599F3BFC010E1EE62E3
 Decrypted Data: Hi There we can work with this

I then ran the result against the following website (https://www.tools4noobs.com/online_tools/xxtea_decrypt/) and it failed to decrypt the the encrypted data with the password provided.

boseji, 2016-12-06:
Now this is a typical problem, let me explain:
The string that you see is the HEX encoded continuous stream of output. Since the data is coded.
So its actually like this -
C8F1CB5F5E7D3BC5EE2753FFC36F8B7AF96759EEF6973599F3BFC010E1EE62E3
which is actually an array:
{ 0xc8, 0xF1, 0xcb, ..... , 0xE3 }
This is not coded with Base64.

If you try to do the tool: https://www.tools4noobs.com/online_tools/xxtea_encrypt/
With base64 enabled only then you get the correct values. Else it would not work.

Here is my suggestion to help you resolve this :
This is a program that would encode the string to Base64 you can also do the reverse process.
```arduino
#include <xxtea-iot-crypt.h>
#include <rBase64.h>

void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println();
  uint8_t keybuf[] = "Hello Password";
  uint8_t plaintext[] = "Hi There we can work with this";
  uint8_t buffer[200];
  int32_t len = 200, i;
  Serial.print(" Password : ");
  Serial.println((char *)keybuf);
  // Setup the Key - Once
  if(xxtea_setup(keybuf, strlen((char *)keybuf)) != XXTEA_STATUS_SUCCESS)
  {
    Serial.println(" Assignment Failed!");
    return;
  }
  Serial.print(" Plain Text: ");
  Serial.println((char *)plaintext);
  // Perform Encryption on the Data
  len = 200;  // - Initialize the Maximum buffer length
  if(xxtea_encrypt(plaintext, strlen((char*)plaintext), buffer, &len) !=
    XXTEA_STATUS_SUCCESS)
  {
    Serial.println(" Encryption Failed!");
    return;
  }
  else
  {
    size_t olen;
    char obuf[400];
    Serial.print(" Encrypted Data: ");
    // Base64 encode
    olen = rbase64_encode(obuf, (char *)buffer, len);
    Serial.println(obuf);
  }
  // Perform Decryption
  if(xxtea_decrypt(buffer, len) != XXTEA_STATUS_SUCCESS)
  {
    Serial.println(" Decryption Failed!");
    return;
  }
  else
  {
    Serial.print(" Decrypted Data: ");
    Serial.println((char *)buffer);
  }
  delay(1000);
}
```
Tested this code on a Arduino Nano so not sure of you too have the same platform.

Try out the Encode Decode now it works.
It uses one more of my libs https://github.com/boseji/rBASE64 in raw mode.

But since XXTEA work on 32bit data-set so some padding is added by default.
You might observe the same in the https://www.tools4noobs.com/online_tools/xxtea_decrypt/

I would try to fix this in the next release, with another API to wrap the older one.

boseji, 2016-12-10:
Let me know in case you have some further queries, hope that the above programs solves your issue.

closed 2016-12-10 by boseji