ardlib/bosejis_xxtea #4: Decrypt string without encrypting first

ThePaceMaker, 2017-07-19:
Hi boseji

From the examples that you have posted, the strings always gets encrypted first before being decryption happens, it then load it into a buffer for decryption.

My problem is, I have a string that returns from a socket that needs decryption. I have the key, but the string is encrypted on the server side. Is there a way just to use just the decryption method without pre encrypting? 

Hope that makes sense.

ThePaceMaker, 2017-07-19:
Please see original post. I have updated it.

boseji, 2017-07-19:
This is similar to the earlier problems. 
Please do not use "https://www.tools4noobs.com/online_tools/xxtea_encrypt/"

They have a completely diffrent way of encryption.
The method used in this library is based around a 80byte or 640bit boundary.
It does not follow the custom chunking done in the mentioned website.

Please use the Golang script provided in the documentation.

ThePaceMaker, 2017-07-20:
Hi boseji

Do you know of a server side library that works with your library, besides the Golang that is provided in the documentation?

boseji, 2017-07-20:
Sorry I don't know currently any one using this encryption.
Use Golang server it works well with the native Crypto wrappers.
Its all about the data width. May be you can modify this code to suit your requirement.
Its open source under CC-BY-SA, please use freely.

boseji, 2017-07-20:
Looks like you had some things earlier too https://github.com/boseji/xxtea-iot-crypt/issues/2
So closing this one also.

closed 2017-07-20 by boseji