ardlib/bosejis_xxtea #6: Crypting for LoRa sx1276

BlackxRyan, 2018-03-20:
Hi.

First, thanks for the library, the code is not too complex and it works well.

I want to use it to secure communication between two Heltec WiFi LoRa 32 ESP32 + OLED modules, a sender and a receiver. These use a technology simlar to the phones. For the moment, I'm sending simple messages.

http://esp32.net/images/Heltec/WIFI-LoRa-32/Heltec_WIFI-LoRa-32_PhotoTop3.jpg

Using the RAW and LoRa exemples, I made a code but I doesn't decrypt the data. In fact, I don't know how to order the commands.

LoRa Sender
```#include <SPI.h>
#include <LoRa.h>
#include <xxtea-iot-crypt.h>

// Pin definetion of WIFI LoRa 32
// HelTec AutoMation 2017 support@heltec.cn 
#define SCK     5    // GPIO5  -- SX127x's SCK
#define MISO    19   // GPIO19 -- SX127x's MISO
#define MOSI    27   // GPIO27 -- SX127x's MOSI
#define SS      18   // GPIO18 -- SX127x's CS
#define RST     14   // GPIO14 -- SX127x's RESET
#define DI0     26   // GPIO26 -- SX127x's IRQ(Interrupt Request)

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

int counter = 0;

String message;

void setup() {
  pinMode(25,OUTPUT);
  
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Sender");

  SPI.begin(SCK,MISO,MOSI,SS);
  LoRa.setPins(SS,RST,DI0);
  LoRa.begin(BAND,PABOOST);
}

void loop() {
  
  uint8_t keybuf[] = "password";
  uint8_t plaintext[] = "Hello";
  uint8_t buffer[200];
  size_t len = 200, i;

  // Setup the Key - Once
  if(xxtea_setup(keybuf, strlen((char *)keybuf)) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Assignment Failed!");
    return;
  }

  // Perform Encryption on the Data
  len = 200;  // - Initialize the Maximum buffer length
  if(xxtea_encrypt(plaintext, strlen((char*)plaintext), buffer, &len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Encryption Failed!");
    return;
  } else {
    Serial.println(" Encrypted Data");
    for(i = 0;i<len;i++)
      message = String (buffer[i], HEX);
    Serial.println();
  }
  
  Serial.print("Sending packet: ");
  Serial.println(counter);
  // send packet
  LoRa.beginPacket();
  LoRa.print(message);
  LoRa.print(counter);
  LoRa.endPacket();
  
  counter++;

  delay(2000);
}
```

LoRa Receiver
```
#include <SPI.h>
#include <LoRa.h>
#include <xxtea-iot-crypt.h>

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

String message;

void setup() {
  Serial.begin(115200);
  while (!Serial); //test this program,you must connect this board to a computer
  Serial.println("LoRa Receiver");
  LoRa.begin(BAND,PABOOST );
}

void loop() {
  uint8_t keybuf[] = "password";
  uint8_t buffer[200];
  size_t len = 200, i;

  xxtea_setup(keybuf, strlen((char *)keybuf));
  
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    while (LoRa.available()) {
      message = (char)LoRa.read();
      xxtea_decrypt(buffer, len);
      Serial.print((char *)buffer);
    }
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}

```

boseji, 2018-03-20:
***Great work !***

Its nice to see the library has been useful to you.

Yes there is particular way in which the bit streams need to be encoded.

Let me point you to the exact portion of the code.

https://github.com/boseji/xxtea-iot-crypt/blob/master/src/xxtea-iot-crypt.cpp#L97

This is where the Low-level function actually encrypts the package.

Now to understand this lets look at an Array of bytes and how the Encryption algorithm handles it:
```c
uint8_t arr[8] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
```

Now when we pass it to the Encryption Function it becomes:
```c
uint32_t arr[2] = { 0x04030201, 0x08070605};
```

This is what the line above does:
```c
memcpy((void *)xxtea_data, (const void *)data, len);
```
This copies the Byte array into a 32bit data array.

You can manipulate this part of the Library to find out how the encoding is going on.

Also one more suggestion from my side would be - encode the data to Base64 using another arduino library. 

That might help to avoid the encoding mismatch. Since after encryption the data becomes binary and can't be handled as character array any more.

I am myself trying to solve this problem since PHP implementation does have issues due to this non-aligned conversion.

Hope this helps.

BlackxRyan, 2018-03-20:
Thanks for this quick answer. That might be just what I needed.

I'm going to study that and I'll keep you informed.

BlackxRyan, 2018-03-21:
I made it with your other library. It works fine and I finally find a way to stock the LoRa.read() chars in a string. I declare "char string[32];" and I use "string[i] = (char)LoRa.read();" in the "for" to stock it.

I just need to know where is the encryption key in your library and if I can modify it to complete my program.

Here's the code :

LoRa Sender

```
#include <LoRa.h>
#include <rBase64.h>

rBase64generic<250> mybase64;

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

int counter = 0;

String message;

void setup() {
  pinMode(25,OUTPUT);
  
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Sender");
  
  LoRa.begin(BAND,PABOOST);
}

void loop() {

  mybase64.encode("Hello");
    
  Serial.print("Sending packet: ");
  Serial.println(counter);
  
  LoRa.beginPacket();
  LoRa.print(mybase64.result());
  LoRa.endPacket();
  
  counter++;

  delay(2000);
}
```

LoRa Receiver
```
#include <LoRa.h>
#include <rBase64.h>

rBase64generic<250> mybase64;

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

char string[32];

void setup() {
  Serial.begin(115200);

  Serial.println("LoRa Receiver");
  LoRa.begin(BAND,PABOOST );
}

void loop() {
  
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    for (int i = 0; i < packetSize; i++)
  {    
     string[i] = (char)LoRa.read();
  }
    mybase64.decode(string);
    Serial.println(mybase64.result());
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}
```

boseji, 2018-03-21:
***Great Work!***

However I could not find where the XXTEA is being used.
Since base64 is only encoding not encryption.

Thanks for mentioning the [rBase64](https://github.com/boseji/rbase64), I got that right this time.
Yes it definitely helps to encode and send out the data.
Since I figured out the String allocation isn't actually helping.

So I had the *rBase64* api changed completely. To make it work.

May be I need to provide some thing similar for the XXTEA also.
It would nice to convert the `class xxtea_c` to a bounded generic C++ . 
However my new changes are going to be a separate class of its own.
Since you are using this library, I will keep the Core API intact and provide a separate generic class in the next update.

Hope that I can close the issue now.

Let me know.

BlackxRyan, 2018-03-21:
I tried to assiociate this code with the xxtea library, but it doesn't work.

LoRa Sender xxtea (64)
```
#include <LoRa.h>
#include <xxtea-iot-crypt.h>
#include <rBase64.h>

rBase64generic<250> mybase64;

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

int counter = 0;

char cryptedtext[32];
char string[32];

String message;

void setup() {
  pinMode(25,OUTPUT);
  
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Sender");
  
  LoRa.begin(BAND,PABOOST);
}

void loop() {
  
  uint8_t keybuf[] = "Abprod";
  uint8_t plaintext[] = "Bonjour";
  uint8_t buffer[200];
  size_t len = 200, i;
  
  // Setup the Key - Once
  if(xxtea_setup(keybuf, strlen((char *)keybuf)) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Assignment Failed!");
    return;
  }

  // Perform Encryption on the Data
  len = 200;  // - Initialize the Maximum buffer length
  if(xxtea_encrypt(plaintext, strlen((char*)plaintext), buffer, &len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Encryption Failed!");
    return;
  } else {
    for(i = 0;i<len;i++)
       cryptedtext[i] = ((char)buffer[i], HEX);
    Serial.println();
  }
  
  mybase64.encode(cryptedtext);
    
  Serial.print("Sending packet: ");
  Serial.println(counter);

  LoRa.beginPacket();
  LoRa.print(mybase64.result());
  LoRa.endPacket();
  
  counter++;

  delay(2000);
}
```

LoRa Receiver xxtea (64)
```
#include <LoRa.h>
#include <xxtea-iot-crypt.h>
#include <rBase64.h>

rBase64generic<250> mybase64;

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

String message;

char string[32];

void setup() {
  Serial.begin(115200);
  while (!Serial); //test this program,you must connect this board to a computer
  Serial.println("LoRa Receiver");
  LoRa.begin(BAND,PABOOST );
}

void loop() {
  uint8_t keybuf[] = "Abprod";
  uint8_t buffer[200];
  size_t len = 200, j;

  xxtea_setup(keybuf, strlen((char *)keybuf));
  
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    for (int i = 0; i < packetSize; i++)
  {    
     string[i] = (char)LoRa.read();
     mybase64.decode(string); 
     for(int j = 0;j<len;j++)
      Serial.print(buffer[j], HEX);
    Serial.println();     
  }
  if(xxtea_decrypt(buffer, len))
      {
        Serial.print((char *)buffer);     
      }
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}
```

boseji, 2018-03-21:
Can you detail whats failing ?

Its not possible to replicate lora communications on my side.
Best would be simpler example where two arduinos are talking to each other.

BlackxRyan, 2018-03-21:
It prints this : 

[image]

BlackxRyan, 2018-03-21:
Here's the simple exemples. You can find the LoRa library here : https://github.com/sandeepmistry/arduino-LoRa

LoRa Sender
```
#include <SPI.h>
#include <LoRa.h>

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true

int counter = 0;

void setup() {
  pinMode(25,OUTPUT);
  
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Sender");
  
  LoRa.begin(BAND,PABOOST);
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);
  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();
  
  counter++;
}
```

LoRa Receiver
```
#include <SPI.h>
#include <LoRa.h>

#define BAND    868E6  //you can set band here directly,e.g. 868E6,915E6
#define PABOOST true
void setup() {
  Serial.begin(115200);
  while (!Serial); //test this program,you must connect this board to a computer
  Serial.println("LoRa Receiver");
  LoRa.begin(BAND,PABOOST );
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}
```

boseji, 2018-03-21:
I am sorry we are going off-topic here.
This library can't help you with your specific application.
Having said that, I would request you to send the encrypted and decrypted output from the XXTEA library.
Only upon that I can divine what going wrong in the XXTEA.

BlackxRyan, 2018-03-21:
No problem. You already helped me a lot, thanks.

closed 2018-03-21 by boseji