ardlib/bosejis_xxtea #8: using on arduino-lora

halukmy, 2018-06-17:
can we use this algorythim in lora?

boseji, 2018-06-18:
Hello,
Its not very safe to be used for physical layer encryption.
However for high level data encryption in packets can be a useful.
Its up to you to use this.
We are presently working to fix a memory buffering issue.
Once that is fixed any amount of data can be encoded.
Also we are trying to make the data ordering change possible, that would help higher level server side processing.

Hope these answer your questions.

halukmy, 2018-06-18:
DearAbhijit,

thanks for nice message and wonderful repo.

Can we work together on this Lora Communication? i working at smart cities 

https://github.com/sandeepmistry/arduino-LoRa using this repo for lora communication

void sendMessage(String outgoing) {
  LoRa.beginPacket();                   // start packet
  LoRa.write(destination);              // add destination address
  LoRa.write(localAddress);             // add sender address
  LoRa.write(msgCount);                 // add message ID
  LoRa.write(outgoing.length());        // add payload length
  LoRa.print(outgoing);                 // add payload
  LoRa.endPacket();                     // finish packet and send it
  msgCount++;                           // increment message ID
}

void onReceive(int packetSize) {
  if (packetSize == 0) return;          // if there's no packet, return

  // read packet header bytes:
  int recipient = LoRa.read();          // recipient address
  byte sender = LoRa.read();            // sender address
  byte incomingMsgId = LoRa.read();     // incoming msg ID
  byte incomingLength = LoRa.read();    // incoming msg length

  String incoming = "";                 // payload of packet

  while (LoRa.available()) {            // can't use readString() in callback, so
    incoming += (char)LoRa.read();      // add bytes one by one
  }

these are send & recieve titles.. 

so i been working and thinking hard to crypt this repo with your nice encryption algorythm

but i couldnt find solution yet.

because its an open communication, so anyone can sniff messages on air,

for example if i send crypted message like     "A@3Qe#" and lets think its an "start" message to start any device on other node 

if someone sniff and copy "A@3Qe#" message and publish on air, probably same code will work on other node..

so how can we secure our wireless rf communication message with your nice repo?

i want to respect and thanks again your message

boseji, 2018-06-18:
Thanks for your kind words,

Actually Lora physical layer is encrypted AES-256 / 128 based on service provider.
Without encryption it would difficult to match it with the Gateway front end network.
Please check with your gateway provider if XXTEA-V2 is alright for them to push to your server.
That would help to get clarity.
Generally there are 3 levels of encryption: Network Key, Application key and High Level Key.
All are AES 128/256 so no need to worry your data would be safe till it reaches your server.

Due to non-availability of specified hardware it would be difficult for us to contribute.
https://github.com/sandeepmistry/arduino-LoRa
Additionally there are many other projects that our time has been allocated with.
We would love to help you at your initiative but its just limited by time and availability.

Thanks again for your invitation.

halukmy, 2018-06-18:
i want to share your project and crypt data solutions in Turkey :)

lorawan really so strong security, but  which i send you repo is open to get message from anyone 

how can we be security that repo with xxtea? because this repo is have not got any  app, network or session key..

only traditional RF Reciever and sender :)

the gateway is also a node,, i just made esp32 + node as gateway module ..  recieve mqtt data and send via open lora message (RF) to other nodes,

also other nodes recieve that node (via esp32) and sent via mqtt message..

i just thinkink how we can secure these rf message with your beautiful security code..

is it possible? :)

halukmy, 2018-06-19:
@boseji  any update? :)

boseji, 2018-06-20:
We welcome your noble initiative.

If we understand correctly all that you would like to do is a -Its an direct radio communication.
Just that the Point to Point communication.
It is then straight forward to do message encoding an Decoding.
One important consideration would be that 80bytes buffer.
If that limitation is alright for you then this library would help.

Here is a mock example to get started with the encryption & decryption:
```c
#include <xxtea-iot-crypt.h>

void setup() {
  uint8_t Password[] = "encryption-password";
  uint8_t Data[] = { 0x02, 0xA4, 0x43, 0x5A, 0xDF, 0xF4, 0xE2, 0xA3 };
  uint8_t txbuffer[200];
  size_t len = 200, i;
  
  // Initialize the Serial Port
  Serial.begin(115200);

  Serial.println();
  
  // Assign Password to the System
  if(xxtea_setup(Password, strlen((char *)Password)) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Assignment Failed!");
    return;
  }

  // Perform Encryption
  if(xxtea_encrypt(Data, strlen((char*)Data), txbuffer, &len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Encryption Failed!");
    return;
  } else {
    Serial.println(" Encrypted Data: ");
    // Transmit the Encrypted Data over Serial port
    for(i = 0;i<len;i++)
      Serial.print(txbuffer [i], HEX);
    Serial.println();
  }

  // Perform Decryption
  if(xxtea_decrypt(txbuffer, len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Decryption Failed!");
    return;
  } else {
    Serial.print(" Decrypted Data: ");
    // Received Data - for Decryption
    Serial.println((char *)txbuffer);
  }
}

void loop() {

}
```

The password once set can be used for Encryption or Decryption until it needs to be changed.
An additional point would be all the Buffer sizes must be in multiples of 4Bytes. 
Else there would conflict.

Hope that this can be your starting point.

closed 2018-06-20 by boseji

halukmy, 2018-06-20:
@boseji every message payload will be different?

boseji, 2018-06-21:
Its upto your implementation.
You can change the input plane text buffer any number of times.
The Password key on either sides needs to be configured only on startup.