ardlib/bosejis_xxtea #14: Base64 string decryption not working

andreddosantos, 2018-11-12:
Hello @boseji

I'm not being able to decrypt messages from my local webserver inside the arduino. With some online tools i've been able to check that the message is properly encrypted
message ->  'message received'
encrypted -> "u+iRmMMTW+EwaFfAjiCrFA==" 
key -> "8b78b1c92031cb91"

But when trying to decrypt inside the microcontroller i have these strange chars as result

>
> Encrypted data: u+iRmMMTW+EwaFfAjiCrFA==
> Decoded data: ⸮葘⸮[⸮0hW⸮⸮ ⸮
> Decrypted Message data: &⸮c⸮⸮N
> 

```
String data = root[String("data")];  //getting encrypted data from JSON
rbase64.decode(data); 
String base64Decoded = rbase64.result();
Serial.println("Decoded data: "+ base64Decoded);
    
String decryptedMessage = xxtea.decrypt(base64Decoded);
Serial.println("Decrypted Message data: "+ decryptedMessage);
```
Can you help me understand what i'm doing wrong?

Regards 
André

boseji, 2018-11-12:
Hi,

Would it be possible to share a bit more detail.
I am not sure if I understand properly your program:
`Input Message=>Encrypt=>Base64-Encode=>Base64-Decode=>Decrypt=>Message`
Let me know if the above is the sequence that you are following.
Or there is some thing different.

Regards,
boseji

boseji, 2018-11-12:
Here is one example based on my understanding:
```c
#include <xxtea-lib.h>
#include <xxtea_core.h>

#include <rBase64.h>

void setup() {
  Serial.begin(115200);

  String data = "[\"message received\"]";  //JSON Data
  Serial.println("Input data: "+ data);
  xxtea.setKey("8b78b1c92031cb91");
  String Encrypted = xxtea.encrypt(data);
  rbase64.encode(Encrypted);
  String base64Encode = rbase64.result();
  Serial.println("Encrypted->Encoded data: "+ base64Encode);   
 //Transmit Encoded Data

  // Received Encoded Data
  rbase64.decode(base64Encode);   
  String base64Decoded = rbase64.result();
  xxtea.setKey("8b78b1c92031cb91");
  String Decrypted = xxtea.decrypt(base64Decoded);
  Serial.println("Decoded->Decrypted data: "+ Decrypted);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

andreddosantos, 2018-11-12:
Yup in this case is the decryption part
On the server i've encrypted the following message with this package [tea npm](https://www.npmjs.com/package/tea/v/0.0.13)

> message -> 'message received'
> encrypted output -> "u+iRmMMTW+EwaFfAjiCrFA=="
> used cripto key -> "8b78b1c92031cb91"

On the microcontroller:
ReceivedMessage_base64 -> decode -> decrypt 

```
String data = root[String("data")];  //getting encrypted data from JSON
rbase64.decode(data); 
String base64Decoded = rbase64.result();
String decryptedMessage = xxtea.decrypt(base64Decoded);
```
and the output for the xxtea.decrypt is some strange chars. But looking at the example provided you set the key again for decryption, in my case i've set the key on setup and nowhere else. Could that key issue be the cause of this strange chars output?

boseji, 2018-11-12:
Do not use `tea npm` this library does not support that.
Encoding format is of xxtea-lib is completely different.
We only support Golang library https://github.com/hillu/go-xxtea
Other implementation will not work.

If you want alternative implementation you can check the following:
https://github.com/xxtea
But again this library will not work with any of the other xxtea implementations that do not follow the exact algo.

closed 2018-11-15 by boseji