ardlib/bosejis_xxtea #16: How to change the output of encrypted text to Base64?

EuphoriaCelestial, 2019-04-20:
I tried your example code, it work fine, but the encrypted text is in HEXA (I guess)
My server side can only work with Base64, how can I change the output to Base64?

[image]

VisualSanity, 2019-04-20:
You can Base64 the text on the IoT device,  then encrypt the Base64 with XXTEA  and then of the server decrypt it back into Base64?

EuphoriaCelestial, 2019-04-20:
No, it not work like that. The input is String here, but it convert to hexa in the library

VisualSanity, 2019-04-21:
The String object is just a wrapper object around a char array. I would try
and stay away from the String object  and just work with the base char
stream.

boseji, 2019-04-22:
@VisualSanity , you are right. Some times the repeated allocation in MCUs can cause fragmentation and corrupt the memory. Its better to use use the Native interface `xxtea`.

@EuphoriaCelestial Try to look at the Raw example :
https://github.com/boseji/xxtea-lib/tree/master/examples/xxtea_Raw

Hope you can get the Char Array that's binary and then use a Base64 converter to generate your stream.
You can also look at another project 
https://github.com/boseji/rBASE64/tree/master/examples/base64-generic
For how to use a Base64 converter.

closed 2019-05-01 by boseji