ardlib/bosejis_xxtea #18: How increase 80 byte limit in Arduino IDE, as there is not xxtea_lib.h file in the installed library?

Alex-7, 2019-11-24:
Hello, 
It is not a bug. The xxtea-lib works fine with my Arduino.

I would like, however, to increase 80byte limitation. But I cannot find **xxtea_lib.h** file in the Arduino IDE as advised in _"Modification needed increase the 80byte at xxtea_lib.h file at line 40"_ .The istalled library in Arduino IDE does not contain this file for some reason.

Perhaps, I could increase the limitation inside my code? After including the library _#include <xxtea-iot-crypt.h>_ ? It would be even better, so the limit does not change with an upgrade.

boseji, 2019-11-24:
Change is needed at :
https://github.com/boseji/xxtea-lib/blob/master/src/xxtea-lib.h#L40

API 2.0 currently is frozen. You can use direct functions in case you need more bytes:
https://github.com/boseji/xxtea-lib/blob/master/src/xxtea-lib.h#L79
https://github.com/boseji/xxtea-lib/blob/master/src/xxtea-lib.h#L80

closed 2019-11-24 by boseji

Alex-7, 2019-11-24:
There are only two files .h in the library installation in Arduino IDE:  xxtea-iot-crypt.h and xxtea_internal.h , but I cannot find _#define MAX_XXTEA_DATA8  80_ in them.

I can see the line 40 in the file xxtea-lib.h on the Github website. But I cannot find this file in the installed Arduino IDE libraries.

boseji, 2019-11-24:
Please provide detail report as per issue template.

Alex-7, 2019-11-24:
I am sorry. I found the definition in the _**xxtea-iot-crypt.h_ file**:
**#define MAX_XXTEA_DATA8  (MAX_XXTEA_DATA32 * 4)**, however when I change it to:
**#define MAX_XXTEA_DATA8  (MAX_XXTEA_DATA32 * 8)** the encryption stops working on my Arduino Mega. Probably not enough resources on the board.

Alex-7, 2019-11-24:
I solved my problem. I changed **#define MAX_XXTEA_DATA32 20** to
**#define MAX_XXTEA_DATA32 40**
And I did not change this line at all:
**#define MAX_XXTEA_KEY8   (MAX_XXTEA_KEY32 * 4)**
And it is working fine on Arduino Mega.