ardlib/bosejis_xxtea #20: Decryption on other arduino doesn't work

urvishjarvis1, 2021-04-05:
Hi 
This is  my code for sender

`#include <dht.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <xxtea-iot-crypt.h>

RF24 radio(9, 10);
const byte address[6] = "00001";
struct package
{
  float temperature ;
  float humidity ;
  uint8_t cipher[20];
};

package packet;

uint8_t keybuf[] = "jarvis";
uint8_t buffer[20];
size_t len = 20, i;

#define dht_apin A0
dht DHT;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  radio.begin(); 
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MIN);
  radio.stopListening();
  delay(500);
  if(xxtea_setup(keybuf, strlen((char *)keybuf)) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Assignment Failed!");
    return;
  }
  Serial.print("Temp and Humidity Coming on!!\n\n");

  delay(10000);
}

void loop() {
  // put your main code here, to run repeatedly:
  DHT.read11(dht_apin);
  Serial.print(" Humidity = ");
  Serial.print(DHT.humidity);
  Serial.print("% ");
  Serial.print("Temp = ");
  Serial.print(DHT.temperature);
  Serial.print("C ");
  packet.humidity = DHT.humidity;
  packet.temperature = DHT.temperature;
  
  String buf;
  buf += F("Temp:");
  buf += String(packet.temperature,4);
  buf += F("\t Humidity:");
  buf += String(packet.humidity,4);
  Serial.println(buf);
  Serial.println(sizeof(buf));
  char message[buf.length()+1];
  buf.toCharArray(message,20);
  Serial.print("msg");
  Serial.println(message);
  
  len = 20;  // - Initialize the Maximum buffer length
  if(xxtea_encrypt(message, strlen((char*)message), buffer, &len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Encryption Failed!");
  } else {
    
    Serial.println(" Encrypted Data: ");
    for(i = 0;i<(sizeof(buffer)/sizeof(buffer[0]));i++){
      Serial.print(buffer[i], HEX);
      packet.cipher[i]=buffer[i];
    }
    Serial.println();
    Serial.println(i);
    
  }
  if(xxtea_decrypt(packet.cipher, len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Decryption Failed!");
    return;
  } else {
    Serial.print(" Decrypted Data: ");
    Serial.println((char *)packet.cipher);
  }
 
  radio.write(&packet,sizeof(packet));
  delay(5000);
  Serial.print("\n");
}`

heres my receiver code:
`#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>
#include <Wire.h> // Library for I2C communication
#include <LiquidCrystal_I2C.h> // Library for LCD
#include <xxtea-iot-crypt.h>

LiquidCrystal_I2C lcd=LiquidCrystal_I2C(0x27,16,2);
RF24 radio(9, 10); // CE, CSN
uint8_t buffer[20];
size_t len = 20, i;
struct package
{
  float temperature ;
  float humidity ;
  uint8_t cipher[20];
};
package data;
uint8_t keybuf[]= "test";
const byte address[6] = "00001";
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  radio.begin();
  radio.openReadingPipe(0, address);   //Setting the address at which we will receive the data
  radio.setPALevel(RF24_PA_MIN);       //You can set this as minimum or maximum depending on the distance between the transmitter and receiver.
  radio.startListening();   
  lcd.init();
  lcd.backlight();
   
  
  delay(500);

  Serial.print("Temp and Humidity Coming on!!\n\n");

  delay(10000);

}

void loop() {
  // put your main code here, to run repeatedly:
  
  if (radio.available())              //Looking for the data.
  { 
   
    radio.read(&data, sizeof(data));
    Serial.print("Size");  
    Serial.println(sizeof(data));//Reading the data
    Serial.print("buff");
    Serial.println(sizeof(data.cipher));
    Serial.print(data.temperature);
    Serial.print(data.humidity);
    
    
    Serial.print("\n");  
    if(xxtea_setup(keybuf, strlen((char *)keybuf)) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Assignment Failed!");
  }
    for(i = 0;i<(sizeof(data.cipher)/sizeof(data.cipher[0]));i++)
    Serial.print(data.cipher[i], HEX);
    Serial.println();
     if(xxtea_decrypt(data.cipher, len) != XXTEA_STATUS_SUCCESS) {
    Serial.println(" Decryption Failed!");
  } else {
    Serial.print(" Decrypted Data: ");
    Serial.println((char *)data.cipher);
  }
  }
  
  lcd.setCursor(2,0);
  lcd.print("Temperature:");
  lcd.print(data.temperature);
  lcd.print("C");
  lcd.setCursor(2,1);
  lcd.print("Humidity:");
  lcd.print(data.humidity);
}`

Now I am getting the same cipher on both arudino but when I try to decrypt the cipher its fails and prints ramdon char

boseji, 2021-04-06:
Can you share the output prints  of either side ?

boseji, 2021-04-16:
Waiting for your inputs. We would auto close the issue if further input is not provided.

urvishjarvis1, 2021-04-16:
I got it working. I just have to use the base64 conversion

closed 2021-04-16 by urvishjarvis1